Snapmaker/Snapmaker2-Firmware-Issues #13: Feature Request: "homing" button for touch screen

Galindorf, 2023-03-21:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

 My machine commonly ends a project with the print bed pushed all the way to the back, making it difficult for me to reach in to the enclosure to get my project, especially if it has tiny little pieces. I have to go into the "controls" and use the touch screen to tap it forward 10mm at a time.... over and over again. 

### Describe the solution you'd like?

 I would like a quick "homing" button or shortcut added to the touch screen that would make the machine reset itself to default position so I don't have to keep doing it manually. Please! :D

scotthsl, 2023-10-08:
@Galindorf  At the end of a print, the hotbed should stay close to the Y-axis endstop position. Is this not the case with your machine? Can you take a picture of where the hotbed stays at the end of a print?

Galindorf, 2023-10-08:
[image]
[image]

This is how my file was set up in Luban, this is how the bed sits when it's finished. I would love to have it always come all the way forward when a job is done!