Snapmaker/Snapmaker2-Firmware-Issues #15: Feature Request: connect to a hidden wlan / Network

TheChosentwo, 2023-03-21:


ggabbiani, 2024-03-18:
Hi everyone, 

any news on this topic? I need to hide my wifi ssid and without this feature I can't...

Many thanks
Giampiero

TheChosentwo, 2024-03-18:
I get no news to it, since a long time

jaminnash, 2024-03-24:
It would be great if the SM team could add this security feature!

rhorer, 2024-05-23:
This request has been open for two years without anyone assigned to it, no milestones, and no branches or pull requests. I see discussion about this topic in the forums going back four years or more. It seems clear that Snapmaker is not going to implement this feature. My Artisan is in an environment where I don't control the network so their suggestion of "just unhide the SSID long enough to join the network" is a bit insulting. Besides, I see in the comments above that the machine disconnects as soon as the SSID goes hidden again. For those who may like to lecture me about how hiding the SSID doesn't provide any meaningful amount of security, I refer you to my previous comment about how I don't control the network.

It's quite disappointing that Snapmaker doesn't care enough about its customers to make a system which costs so much and is so well-engineered in other areas do something as basic as connect (and stay connected) to a hidden SSID. C'mon guys, this is low-hanging fruit.

elsbrock, 2026-11-05:
Easy workaround (iPhone):

1. **Settings > General**: rename iPhone to name of your hidden wifi
2. **Settings > Personal Hotspot**: verify the name of your hotspot, change the password to that of the hidden wifi and enable "Maximize compatibility". Copy your old password to clipboard to restore easily.
3. Connect the printer via the iPhone's hotspot.
4. Disable the hotspot and restore the password and phone name on the iPhone.

The printer should now automatically connect to your hidden wifi.